tkoecker/OctoPrint-Mmu2filamentselect #1: Add a Cancel button to the popup screen

ThePiGuy, 2021-02-22:
Occasionally, I find I've started a print, and want to cancel it before I've even selected a filament colour. 

It would be great if there was a red cancel button below the 5 filament select choices that allowed me to cancel the print before the MMU has even begun loading the filament, instead of having to reset the printer, or wait while the MMU loads and then unloads the filament after a Cancel in OctoPrint.

tkoecker, 2021-02-26:
I'm not sure how much speedup this will give in comparison to selecting a filament and canceling the print immediately afterwards, but it does not sound to difficult to implement. I will give it a try.

ThePiGuy, 2021-04-19:
Hey, wondering if you've had a chance to look at this at all yet? I just cancelled another print after waiting for the dialog to go away, and remembered how sweet this feature would be. Thanks!

JKinch03, 2021-04-23:
I think all that would need to be done is to issue the code to stop the print midway. I don't believe that this is a Gcode command, and it's just built into the printer's firmware as a feature there... but I might be mistaken. Also, as Prusa hasn't implemented this into the printer itself yet, I'm not 100% sure that it would work.

ThePiGuy, 2021-04-23:
I'm not certain, but I think the button only needs to issue the OctoPrint API command to stop the print; no Gcode needs to be sent. At the point where the selection box is up, the printer is awaiting further instruction from OctoPrint, and is not executing any blocking commands that would need to be halted.

JKinch03, 2021-04-23:
Yeah, that sounds more accurate than what I was thinking would have to happen. It seems pretty simple to do if you know how to edit the code, but I don't, and I don't want to be the one to mess something up....lol. It would be a really nice feature to have though.

tkoecker, 2021-04-23:
Yes, it is an OctoPrint API command. Actually I did a quick attempt to implement it, but did not yet get around testing it / comparing it to just select any random filament and clicking on the regular "cancel print" button right after that (which should afaik be quite similar in regards what would happen).

tkoecker, 2021-05-28:
Added in release 1.0.4

closed 2021-05-28 by tkoecker