tkoecker/OctoPrint-Mmu2filamentselect #4: OctoPrint did not come back following restart after install

evannadeau, 2021-03-16:
Just like the title says, OctoPrint did not come back after the install.

I removed the previous version (from the previous supporter, 1.0.2) and restarted OctoPrint.  I then installed your version (1.0.3), and this was the only item being installed or updated at this time, and performed the restart of OctoPrint service.  However it did not come back.  I should have taken a look at OctoPi via putty, but did not.  But I power cycled my pi (Raspberry Pi 4 Model B Rev 1.1), and it came back up, and showed that the plugin was installed.

I'll be starting a print shortly, and will confirm the plugin is working.

tkoecker, 2021-06-06:
Do you have any new observations, or logs for this issue? If not and its working now I will close this issue.

evannadeau, 2021-06-07:
Feel free to close it. I don’t think this was an issue from your plug-in.

closed 2021-06-07 by tkoecker