tkoecker/OctoPrint-Mmu2filamentselect #5: The choosing menu do not show up

romankysely, 2021-04-07:
Hi,
when I start print, the menu for choosing fillament do not showup. When I can expect menu? Immediately after push Print button or lately after warmup of nozzle and bed?

Thx a lot for help.
Roman

**Configuration:**
MK3S+MMU2S latest FW configured as Single
PrusaSlicer 2.3.0
OctoPrint 1.5.3
OctoPi 0.18
[System Information OctoPrint.txt](https://github.com/tkoecker/OctoPrint-Mmu2filamentselect/files/6274117/System.Information.OctoPrint.txt)

Prusa MMU2 Select Filament: 1.0.3 
GCode file from PrusaSlicer:
[test_gcode.zip](https://github.com/tkoecker/OctoPrint-Mmu2filamentselect/files/6274096/test_gcode.zip)

IzK666, 2021-04-11:
The choosing menu doesn't appears when the file is printed from the SD.
Can you fix it?

romankysely, 2021-04-14:
Thx for reply. I do not print from SD card. I am printing via Octopi. PrusaSlicer send gcode diretly to Octoprint and there I choose file from menu and push Print button. The file with gcode is stored in Octoprint server. Any idea? thx

romankysely, 2021-04-22:
The problem was at profiles of PrusaSlicer 2.3. They did not generate Tx command in GCode.

JKinch03, 2021-04-23:
Hey @romankysely. You might want to check in the Octoprint settings for your printer configuration. I don't remember the exact menu path, but it should be under settings, and you should see something along the lines of an "about your printer" menu. You'll need to change the number of nozzles on the printer to 5 for the plugin to work properly.
Hope this helps!
-JKinch

tkoecker, 2021-06-06:
Yes, the dialog only shows up if the generated gcode contains "Tx"
Closing issue as the cause was found.

closed 2021-06-06 by tkoecker